MaJerle/stm32-cube-cmake-vscode #5: Fonts and Images need updating in example TouchGFX cmake project

edholmes2232, 2022-03-29:
When any fonts are changed in TouchGFX Designer, the default makefile runs `imageconvert` and `textconvert` tools to generate any changes. The example CMakeLists.txt does not run the tools. 

The solution I've found:
`script-projects/h735g-dk-touchgfx/CMakeLists.txt:331`
```C
# Generate TouchGFX Images
add_custom_command(TARGET ${EXECUTABLE}
	PRE_BUILD
	COMMAND ${PROJ_PATH}/Middlewares/ST/touchgfx/framework/tools/imageconvert/build/linux/imageconvert.out -r ${PROJ_PATH}/TouchGFX/assets/images -w ${PROJ_PATH}/TouchGFX/generated/images
	DEPENDS ${PROJ_PATH}/TouchGFX/assets/images
)
# Generate TouchGFX Fonts
add_custom_command( TARGET ${EXECUTABLE}
	PRE_BUILD
	COMMAND ${CMAKE_COMMAND} -E make_directory ${PROJ_PATH}/TouchGFX/generated/texts/include/texts
	COMMAND ruby ${PROJ_PATH}/Middlewares/ST/touchgfx/framework/tools/textconvert/main.rb ${PROJ_PATH}/TouchGFX/assets/texts/texts.xml ${PROJ_PATH}/Middlewares/ST/touchgfx/framework/tools/fontconvert/build/linux/fontconvert.out ${PROJ_PATH}/TouchGFX/generated/fonts ${PROJ_PATH}/TouchGFX/generated/texts ${CMAKE_SOURCE_DIR}/TouchGFX/assets/fonts ${PROJ_PATH}/TouchGFX
	DEPENDS ${PROJ_PATH}/TouchGFX/assets/texts/texts.xml
)
```

Let me know if there is a more efficient way to do this with Cmake.

MaJerle, 2022-03-29:
Isn't TouchGFX in charge to run these commands even before you come to CMake?

mucahitdemir, 2022-05-05:
I wonder the same answer that MaJerle has asked. This website also might help: https://visualgdb.com/tutorials/arm/import/

edholmes2232, 2022-05-05:
No, I am certain TouchGFX does not run these commands. 
Using an existing project, I opened TouchGFX designer (4.18), and modified some text "HELLO WORLD" > "HELLO SAILOR".
Then clicked "Generate Code". Below is the log:

```
Generate Code
    Generate
        Done
    Generate Assets
        make -f simulator/gcc/Makefile assets -j8
        Done
    Post Generate
        touchgfx update_project --project-file=simulator/msvs/Application.vcxproj
        Done
    Post Generate Target
        touchgfx update_project --project-file=../TestProj.ioc --platform=m4
        Done
    Done
```

It is running commands from the default makefile generated when the project was created. The only file changed is `TouchGFX/assets/texts/texts.xml`

Running CMake Clean Rebuild without the above solution still produces a build. Running this build does not change the text. The text shown is "HELLO WORLD".

If the above solution is added to the `CMakeLists.txt` and the project rebuilt, many more files in `TouchGFX/generated` are modified, and the build now shows the new text.

I'm sure a similar example can be demonstrated for Images.

MaJerle, 2022-05-05:
It is not true - when you modify project in TouchGFX designer and generate, it does generate new sources, however it does not update my CMakeLists.txt file. I solved this "problem" in my case, by manually adding TouchGFX sources to the project.

A new `CMakeLists.txt` file is added in the TouchGFX directory:

```
cmake_minimum_required(VERSION 3.22)

# Debug message
message("Entering ${CMAKE_CURRENT_LIST_DIR}/CMakeLists.txt")

#
# Setup project if this is top-level
#
if (PROJECT_IS_TOP_LEVEL)

project(TouchGFX-simulator)
set(EXECUTABLE                      ${CMAKE_PROJECT_NAME})
enable_language(C CXX ASM)

endif()

#
# Setup all necessary include directories
#
set(touchgfx_include_DIRS
    ${CMAKE_CURRENT_LIST_DIR}/Middlewares/ST/touchgfx/framework/include
    ${CMAKE_CURRENT_LIST_DIR}/TouchGFX/generated/fonts/include
    ${CMAKE_CURRENT_LIST_DIR}/TouchGFX/generated/gui_generated/include
    ${CMAKE_CURRENT_LIST_DIR}/TouchGFX/generated/images/include
    ${CMAKE_CURRENT_LIST_DIR}/TouchGFX/generated/texts/include
    ${CMAKE_CURRENT_LIST_DIR}/TouchGFX/generated/videos/include
    ${CMAKE_CURRENT_LIST_DIR}/TouchGFX/gui/include
    ${CMAKE_CURRENT_LIST_DIR}/TouchGFX/App
    ${CMAKE_CURRENT_LIST_DIR}/TouchGFX/target/generated
    ${CMAKE_CURRENT_LIST_DIR}/TouchGFX/target
    ${CMAKE_CURRENT_LIST_DIR}/TouchGFX/simulator/
    )

#
# Scan for all necessary source files to be part of the build
#
file(GLOB_RECURSE graphics_core_generated_SRCS
            TouchGFX/App/*.cpp
            TouchGFX/App/*.c
            TouchGFX/generated/*.cpp
            TouchGFX/gui/*.cpp
            TouchGFX/gui/*.c
            TouchGFX/target/*.cpp
            TouchGFX/simulator/*.cpp
)

#
# Remove simulator files it this is not top-level project
#
if (NOT PROJECT_IS_TOP_LEVEL)
list(FILTER graphics_core_generated_SRCS EXCLUDE REGEX ".*/simulator/.*")
endif()
message("graphics source files: ${graphics_core_generated_SRCS}")

#
# Set executable when project is top-level
#
if (PROJECT_IS_TOP_LEVEL) 

# Add executable and 
add_executable(${EXECUTABLE})
target_sources(${EXECUTABLE}
    ${graphics_core_generated_SRCS})
target_include_directories(${EXECUTABLE} PRIVATE 
    ${touchgfx_include_DIRS})
target_compile_definitions(${EXECUTABLE} PRIVATE
    WIN32
    NDEBUG
    _CONSOLE
)

else()
# Register core library for the system
# This is not top-level library
add_library(touchgfx INTERFACE)
target_sources(touchgfx PUBLIC ${graphics_core_generated_SRCS})
target_include_directories(touchgfx INTERFACE ${touchgfx_include_DIRS})
endif()

# Debug message
message("Exiting ${CMAKE_CURRENT_LIST_DIR}/CMakeLists.txt")

```

And then top level CmakeLists includes it as subdirectory

```
# Add external libraries
add_subdirectory("../relative/path/to/new/cmakelists" touchgfx)
target_link_libraries(${PROJECT_NAME} touchgfx)
```

Each time I update TouchGFX project, I go back to top level CMake file and re-run build to add new files to build - using GLOB.

mucahitdemir, 2022-05-05:
Thanks for sharing this.