MaJerle/stm32-cube-cmake-vscode #7: Advice against the use of cmake-kits, please use cmake's native CMakePresets.json instead

daantimmer, 2022-05-17:
VScode's cmake-kits is, well, vscode specific and does not transfer (well) to a CI environment.

cmake 3.19 introduced [cmake-presets](https://cmake.org/cmake/help/v3.19/manual/cmake-presets.7.html). This is cmake's preferred way of using the same functionality that vscodes cmake-kits provide.

I'll provide an example CMakePresets.json from one of my (pet) projects:

```json
{
    "version": 3,
    "configurePresets": [
        {
            "name": "default",
            "hidden": true,
            "generator": "Ninja",
            "binaryDir": "${sourceDir}/${presetName}",
            "cacheVariables": {
                "CMAKE_EXPORT_COMPILE_COMMANDS": "ON",
                "CMAKE_C_COMPILER_LAUNCHER": "ccache",
                "CMAKE_CXX_COMPILER_LAUNCHER": "ccache"
            }
        },
        {
            "name": "Build-Host",
            "inherits": "default",
            "cacheVariables": {
                "CMAKE_BUILD_TYPE": "Debug"
            }
        },
        {
            "name": "Build-Firmware",
            "inherits": "default",
            "toolchainFile": "${sourceDir}/cmake/toolchain-arm-gcc.cmake",
            "cacheVariables": {
                "CMAKE_BUILD_TYPE": "RelWithDebInfo",
                "emil_DIR": "${sourceDir}/Build-Host/_deps/emil-build/cmake",
                "EMIL_PACKAGE_CONFIG_IMPORT_NAMESPACE": "emil::",
                "CHIP_VENDOR": "ST",
                "CHIP_ST_SERIES": "STM32F4xx",
                "CHIP_ST_FAMILY": "STM32F411xE"
            },
            "warnings": {
                "dev": false
            }
        }
    ]
}
```

As you can see here the toolchain file can be set as well.

I don't currently have a build and test preset because for this project I don't need them.
You can now invoke cmake like (CI example, you can leave out the extra parameters for a local build if you wish):

```
cmake --preset Build-Host -DCMAKE_CXX_COMPILER_LAUNCHER=ccache
cmake --build Build-Host
ctest --test-dir Build-Host --quiet --no-compress-output --output-on-failure
cmake --preset Build-Firmware-DCMAKE_CXX_COMPILER_LAUNCHER=ccache
cmake --build Build-Firmware
```

MaJerle, 2022-05-17:
Thanks for your report.

I tried your code, if I add this code, it already seems to work well. I will consider to add it to converter.py and to update docs.

```json
{
    "version": 3,
    "configurePresets": [
        {
            "name": "default",
            "hidden": true,
            "generator": "Ninja",
            "binaryDir": "${sourceDir}/build/${presetName}/build",
            "toolchainFile": "${sourceDir}/gcc-arm-none-eabi.cmake",
            "cacheVariables": {
                "CMAKE_TOOLCHAIN_FILE": "${sourceDir}/gcc-arm-none-eabi.cmake",
                "CMAKE_EXPORT_COMPILE_COMMANDS": "ON"
            }
        },
        {
            "name": "Debug",
            "inherits": "default",
            "cacheVariables": {
                "CMAKE_BUILD_TYPE": "Debug"
            }
        },
        {
            "name": "Release",
            "inherits": "default",
            "cacheVariables": {
                "CMAKE_BUILD_TYPE": "RelWithDebInfo"
            }
        }
    ]
}
```

This allows user to add more configuration (bootloader, main-app, ...).

MaJerle, 2022-07-06:
I think it is done now.

closed 2022-07-06 by MaJerle

daantimmer, 2022-07-06:
I saw the result, looks good :-)