MaJerle/stm32-cube-cmake-vscode #18: STM32 Cube vscode not supported in Windows 11

TheAfrvlz, 2023-06-04:
with the extensions that the extension mentions, the project doesnt work

MaJerle, 2023-06-05:
Can you elaborate more where does it fail?

TheAfrvlz, 2023-06-05:
[build] 'cmd.exe' is not recognized as an internal or external command, 
[build] operable program or batch file,
[build] ninja: build stopped: subcommand failed." this is the message that vscode shows. I follow all the instructions but doesnt work

MaJerle, 2023-06-05:
This error is not associated with extensions.
Can you run commands inside vscode terminal? Any command

TheAfrvlz, 2023-06-05:
I can run commands, but in the building process, thats the message

MaJerle, 2023-06-05:
Cmake configuration goes through well? Can you give soke log data or something? Ninja, cmake, gcc are all in path?

TheAfrvlz, 2023-06-05:
[proc] The command: c:\Users\Jvele\AppData\Roaming\Code\User\globalStorage\ms-vscode.vscode-embedded-tools\vcpkg\root\downloads\artifacts\vcpkg-ce-default\tools.kitware.cmake\3.20.1\bin\cmake.EXE --build "C:/Users/Jvele/OneDrive/Escritorio/BackUp/CODIGOS/Expert IO Learning/Stm32f4_BaseProject/build/debug/build" exited with code: 1
[build] ninja: build stopped: subcommand failed.
[driver] Build completed: 00:00:02.007
[build] Build finished with exit code 1  

cmakelist.txt

cmake_minimum_required(VERSION 3.20)

project("Stm32f4_BaseProject" C CXX ASM)

include(cmake/st-project.cmake)

vcpkg file
{
  "registries": [
    {
      "name": "microsoft",
      "location": "https://aka.ms/vcpkg-ce-default",
      "kind": "artifact"
    }
  ],
  "requires": {
    "microsoft:tools/kitware/cmake": "* 3.20.1",
    "microsoft:tools/ninja-build/ninja": "* 1.10.2"
  }
}

add_executable(${PROJECT_NAME})
add_st_target_properties(${PROJECT_NAME})

MaJerle, 2023-06-05:
You are using vcpkg. Try to disable embedded tools plugin and test again.

TheAfrvlz, 2023-06-05:
embedded tools can not be disable, mentions that stm32 vscode extension depends on embedded tools

MaJerle, 2023-06-05:
Ok - this repository does not rely on stm32 for vscode nor embedded tools. It only relies on the extensions provided in the recommendation file.

You may have forgot to enable vcpkg system.

And also, are you behind proxy?

MaJerle, 2023-06-05:
Further support shall be handled by Microsoft since they are Embedded Tools provider.

closed 2023-06-05 by MaJerle

robotdad, 2023-06-05:
@TheAfrvlz you can find the issue list for embedded tools extension here: https://github.com/microsoft/vscode-embedded-tools/issues

TheAfrvlz, 2023-06-06:
I enable vcpkg and still doesnt work the compilation. You should make a page for installation better than your video. https://www.youtube.com/watch?v=guNg5OVhakU,

Norlander, 2023-06-08:
@TheAfrvlz, Are you relying on STM32CubeCLT tools or STM32CubeIDE to source toolchain debugger etc?
If you are relying in CubeCLT try to have a look inside:
_cmake/gcc-arm-noen-eabi-cmake:_
There is a line pointing to the CLT installation:
`set(WINDOWS_ST_CLT_PATH "C:/ST/STM32CubeCLT/STM32CubeCLT/GNU-tools-for-STM32/bin/")`

For me it looked like that which was not correct on my computer. I corrected this path and then it worked.

What triggered the problem I found to be inside: Rules.ninja:

`rule C_EXECUTABLE_LINKER__VSCode_F407_test_Debug
  command = cmd.exe /C "$PRE_LINK && C:\ST\STM32CubeCLT\STM32CubeCLT\GNU-tools-for-STM32\bin\\arm-none-eabi-gcc.exe -fdata-sections -ffunction-sections --specs=nano.specs -Wl,--gc-sections $FLAGS $LINK_FLAGS $in -o $TARGET_FILE $LINK_PATH $LINK_LIBRARIES && $POST_BUILD"
  description = Linking C executable $TARGET_FILE
  restat = $RESTAT
`

I removed the ` && $POST_BUILD` and that also removed the symptom but did not cure the disease which was the incorrect path.