MaJerle/stm32-cube-cmake-vscode #20: CMakeLists.txt with Internal Variables Fails

MucaGinger, 2023-07-05:
I was following original post _stm32-cube-cmake-vscode_ where basic 'CMakeLists.txt' and 'gcc-arm-none-eabi.cmake' files are introducted and explained. However, if I try generating CMake from files given under 'template' folder, I get bunch of errors referring to 'sr: ...' variables. This is the output log I get after running 'cmake --preset Debug':

'
Preset CMake variables:

  CMAKE_BUILD_TYPE="Debug"
  CMAKE_EXPORT_COMPILE_COMMANDS="ON"
  CMAKE_TOOLCHAIN_FILE:FILEPATH="C:/Users/Luka/Desktop/TEST/cmake/gcc-arm-none-eabi.cmake"

-- The C compiler identification is GNU 10.3.1
-- The CXX compiler identification is GNU 10.3.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/ST/STM32CubeIDE_1.12.0/STM32CubeIDE/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.10.3-2021.10.win32_1.0.200.202301161003/tools/bin/arm-none-eabi-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/ST/STM32CubeIDE_1.12.0/STM32CubeIDE/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.10.3-2021.10.win32_1.0.200.202301161003/tools/bin/arm-none-eabi-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- The ASM compiler identification is GNU
-- Found assembler: C:/ST/STM32CubeIDE_1.12.0/STM32CubeIDE/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.10.3-2021.10.win32_1.0.200.202301161003/tools/bin/arm-none-eabi-gcc.exe
Build type: Debug
CMake Error at CMakeLists.txt:72 (add_executable):
  The target name "{{sr:project_name}}" is reserved or not valid for certain
  CMake features, such as generator expressions, and may result in undefined
  behavior.

CMake Error at CMakeLists.txt:75 (target_sources):
  Cannot specify sources for target "{{sr:project_name}}" which is not built
  by this project.

CMake Error at CMakeLists.txt:78 (target_include_directories):
  Cannot specify include directories for target "{{sr:project_name}}" which
  is not built by this project.

CMake Error at CMakeLists.txt:86 (target_compile_definitions):
  Cannot specify compile definitions for target "{{sr:project_name}}" which
  is not built by this project.

CMake Error at CMakeLists.txt:98 (target_link_libraries):
  Cannot specify link libraries for target "{{sr:project_name}}" which is not
  built by this project.

-- Configuring incomplete, errors occurred!
'

I modified  'CMakeLists.txt' and 'gcc-arm-none-eabi.cmake' files only with paths and some flags so I assume it should work. Unless I misunderstood some other part of the process.

[gcc-arm-none-eabi.txt](https://github.com/MaJerle/stm32-cube-cmake-vscode/files/11960984/gcc-arm-none-eabi.txt)
[CMakeLists.txt](https://github.com/MaJerle/stm32-cube-cmake-vscode/files/11960969/CMakeLists.txt)

MaJerle, 2023-07-05:
Can you sbow me full path of your cubemx/cubeide project, includinf path on a disk, and a command you have invoked?

MaJerle, 2023-07-06:
So when you generated project from CubeMX, which IDE have you selected? It must be STM32CubeIDE.
Can you show the response from python converter script?

Again, template folder is not meant for user, it is mean for converter.py script, that does automatic project conversion from STM32CubeIDE-based project to CMake.

You can get STM32CubeIDE-based project with 2 options:
- Use STM32CubeMX and select STM32CubeIDE as IDE when generating the code
- Actually use STM32CubeIDE to generate the project

MucaGinger, 2023-07-06:
True, I'm building my STM32 project as Makefile project from CubeMX. If I would use STM32CubeIDE then I would probably have to take additional step of re-generating files from CubeIDE like you have shown in your vscode guide.

Then, I guess I will be using the basic template :)

MaJerle, 2023-07-06:
If you use Makefile project (which is obsolete, btw), then why do you even use CMake? You anyway need to do "manual" work.

If you use CubeMX/CubeIDE, make sure project is generated as CubeIDE project and you run the auto-script, that takes care of everything. I guess we can close the issue.

closed 2023-07-06 by MaJerle