MaJerle/stm32-cube-cmake-vscode #21: No Cortex Registers view and Disassembly View

MucaGinger, 2023-07-06:
I just finished your tutorial and most of the stuff works as should. Only disassembly view and cortex registers view (in my case probably cortex live watch) aren't "operational" yet.

When searching for disassembly view from command palette, I get this:
[image]
However I managed to find some other form of memory view (probably default vscode memory view?) but it doesn't look like it is the one `cortex-debug` extension is providing. This is what I get:
[image]

And for cortex register view I subsequently enabled `liveWatch` property in `launch.json` file as the hint says, however no changes to current setup. As far as I know I should be able to see core registers in that section? Well, this is what I get and my `launch.json` file:
[image]

MucaGinger, 2023-07-06:
It seems Cortex Registers View was moved to Variable View as one of the available drop-down lists. And for Disassembly view, it seems I cannot open it like you could normally do (or right-clicking code and opening Disassembly View from there - both aren't available for me). I find the other Disassembly view from your tutorial a bit more "aesthetic looking", however I can probably get along the existing one as well.

MaJerle, 2023-07-06:
- Disassembly view provides nice view of the C code and assembly next to that -> view that you have shown
- Cortex-Live-watch is a generic variable live watch. It has nothing to do with cortex registers. You have to add variable

All looks ok.

closed 2023-07-06 by MaJerle